MeteoSwiss/pyrad #11: Adapt Pyrad to Python 3.9

jfigui, 2021-09-28:
Currently Pyrad is not compatible with Python 3.9. We have to investigate further the issue.

wolfidan, 2021-09-28:
The log of the error is 

ERROR: Command errored out with exit status 1: /usr/share/miniconda/envs/test-environment/bin/python3.9 -u -c 'import io, os, sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-f7dzknbw/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-f7dzknbw/setup.py'"'"';f = getattr(tokenize, '"'"'open'"'"', open)(__file__) if os.path.exists(__file__) else io.StringIO('"'"'from setuptools import setup; setup()'"'"');code = f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-d54ycjmp/install-record.txt --single-version-externally-managed --compile --install-headers /usr/share/miniconda/envs/test-environment/include/python3.9/pyart-mch Check the logs for full command output.
Error: Process completed with exit code 1.

So it fails during C compilation of pyart I think. Unfortunately I don't know where to access the "logs for full command output". They don't seem to be listed in the github actions page...

wolfidan, 2022-03-15:
This was done, python3.9 is now supported

closed 2022-03-15 by wolfidan