MeteoSwiss/pyrad #15: Make "MeteoSwiss" the owner of pyrad-mch on pypi

fpavogt, 2022-02-18:
Dear pyrad devs,

To improve the handling and dissemination of MCH Python codes on pypi and testpypi, a new ["MeteoSwiss"](https://pypi.org/user/MeteoSwiss/) account has been setup on both platforms. 

The idea would be for "MeteoSwiss" to be the sole "pypi owner" of all codes on there, including pyrad-mch. The current owner(s) would become "maintainer", such that they would still be able to upload new releases without issues. Doing so requires the current pypi owner to login, and on the code settings' page, add "MeteoSwiss" as a collaborator and owner of the code. 

Do you see any problems doing so ? 

Incidentally, API tokens for pypi and testpypi have been stored as Github Secrets in this organization. These can be used to automate the upload of the code to pypi/testpypi via Github Actions, if you are so inclined. For example, [like this](https://github.com/MeteoSwiss/ampycloud/blob/develop/.github/workflows/CI_pypi.yml).

wolfidan, 2022-02-28:
Dear [fpavogt](https://github.com/fpavogt),
That's a good idea, I sent you an invitation from pypi,

Kind regards,
Daniel

fpavogt, 2022-03-01:
Hi @wolfidan, 

Great ! I've just accepted the pypi invite, and made `meteoswiss-mdr` the package maintainer. You should still be able to upload new releases as you used to up-to-now.

wolfidan, 2022-03-01:
Thank you! I'm closing the issue,

closed 2022-03-01 by wolfidan