MeteoSwiss/pyrad #25: bug in plotting colorbar in plot_grid when field has norm

jfigui, 2022-10-21:
When I tried to plot a field with norm I get this:
[image]
This did not happen before we aligned it with the ARM-DOE pyart. I suspect that it has to do with using the wrong axis but I got lost. @wolfidan , can you have a look at it?

wolfidan, 2022-10-26:
Ciao @jfigui ,

I have some trouble reproducing your issue, would you mind sharing your pyrad config file?
Cheers

jfigui, 2022-10-27:
@wolfidan 
[mf_bin_qpe_toul.txt](https://github.com/MeteoSwiss/pyrad/files/9875893/mf_bin_qpe_toul.txt)
[mf_bin_qpe_toul_loc.txt](https://github.com/MeteoSwiss/pyrad/files/9875894/mf_bin_qpe_toul_loc.txt)
[mf_bin_qpe_toul_prod.txt](https://github.com/MeteoSwiss/pyrad/files/9875895/mf_bin_qpe_toul_prod.txt)
I will send you an example data file by mail.

Cheers,

Jordi

wolfidan, 2022-12-01:
Hi @jfigui 

The issue occurs [here](https://github.com/MeteoSwiss/pyrad/blob/master/src/pyrad_proc/pyrad/graph/plots_aux.py#L305). Why do we use cmap.N as the number of colors, to me it makes no sense to use more colors than levels. Setting ncolors = len(boundaries) -1 seems to fix the problem for me.
Would that be an option?

jfigui, 2022-12-02:
Hi @wolfidan ,

I would have never guessed that was the issue! Well done! Change it in the dev version and we can make sure that nothing breaks before the new release

wolfidan, 2022-12-20:
Solved in https://github.com/MeteoSwiss/pyrad/releases/tag/v1.6

jfigui, 2023-02-16:
We solved the plotting when using xarrays data but we created a new one when using regular fields such as in the PPI or RHI plots

wolfidan, 2023-02-16:
@jfigui 

Sorry about that, I think we need to make it more specific then, such as with a isxarray boolean in the get_norm function, that would be true only when the function is called from plot_grid
I will change that

jfigui, 2023-02-16:
@wolfidan 
Perfectly fine solution. Thanks

wolfidan, 2023-02-16:
I pushed it to the latest commit, unfortunately the tests fail due to a weird xarray issue, I have to look into it

wolfidan, 2023-02-17:
I managed to make the tests pass for 3.8 and 3.9. I don't have the time currently to check why 3.7 fails in the test (a weird cython or xarray issue that does not concern pyrad), so I removed 3.7 from the tests. Could you check if it works with the  latest commit on dev please.