MeteoSwiss/pyrad #34: Why are user specified lat/lon limits in grid exagerated by pyrad

wolfidan, 2023-02-02:
Hi @jfigui

What is the reason for these two lines in plots_grid.py?

````
lon_lines = np.arange(np.floor(min_lon), np.ceil(max_lon)+1, lonstep)
lat_lines = np.arange(np.floor(min_lat), np.ceil(max_lat)+1, latstep)
````

This leads to a domain that is far too big if you are interested in more local features,
for example setting 

lonmin     FLOAT  7.4 # Min longitude [deg]
   lonmax     FLOAT  8.4 # Max longitude [deg]
   latmin     FLOAT 46.4 # Min latitude [deg]
   latmax     FLOAT 47.1 # Max latitude [deg]

in the loc config file will give limits of 46-48.5 and 7.0-9.5 in the plots which is much larger.
Thanks

wolfidan, 2023-02-02:
I personally would change that to 
````
lon_lines = np.arange(min_lon, max_lon+lonstep, lonstep)
lat_lines = np.arange(min_lat, max_lat + latstep, latstep)
````

jfigui, 2023-02-03:
Hi @wolfidan ,
There was no particular reason except that you have to give some marging on the limits. The proposed solution within pyart is this one:

            if lon_lines is None:
                lon_lines = np.linspace(
                    np.around(ds.lon.min()-.1, decimals=2),
                    np.around(ds.lon.max()+.1, decimals=2), 5)

            if lat_lines is None:
                lat_lines = np.linspace(
                    np.around(ds.lat.min()-.1, decimals=2),
                    np.around(ds.lat.max()+.1, decimals=2), 5)

It probably helps avoiding steps that are too small.

wolfidan, 2023-03-30:
Ok I think it makes sense, we can close the issue for now

closed 2023-03-30 by wolfidan