MeteoSwiss/pyrad #40: ODIM data not properly masked

jfigui, 2023-03-31:
The implemented solution for when nodata was np.nan:

data = np.ma.masked_where(raw_data, raw_data == nodata)

was masking all data.

This has been changed in the dev commit b3a1b0f6102143be0049490f4f4dc90f5d4bad8c as:
if np.isnan(nodata):
                    # special case of nan nodata
                    data = np.ma.masked_array(raw_data, np.isnan(raw_data))
                else:
                    data = np.ma.masked_values(raw_data, nodata)

@wolfidan , we need to release a patch as soon as possible to version 1.7

wolfidan, 2023-04-06:
Can I close this issue?

jfigui, 2023-04-11:
Hi @wolfidan ,

Yes, I think that with the patch we can close this issue.

Cheers,

Jordi

closed 2023-04-11 by jfigui