MeteoSwiss/pyrad #41: Support slightly different times in input PYRAD datasets (CFRADIALPYRAD, ODIMPYRAD, GRIDPYRAD)

wolfidan, 2023-04-20:
In my particular case of use (dual-doppler) it would be useful to be able to get pre-processed grids from pyrad from two different radars which have slightly different reference times. In my case the dataset types are GRIDPYRAD but a similar scenario could appear for CFRADIALPYRAD or ODIMPYRAD.

I was wondering if we could add for example a file_time_tolerance (for example) parameter in the loc (or main?) config file. If larger than zero, the code would not look for the exact time of the masterfile but rather the available file that is the closest in time to the masterfile if this time offset is less than file_time_tolerance. This would add some flexibility for multi-radar applications.
@jfigui, what do you think about this possibility?

jfigui, 2023-04-20:
@wolfidan ,
The variable 'TimeTol' already exists and it allows to get data from radars with different nominal times. See _get_radars_data in flow_aux.py.
In addition, for odim files there is the variable 'MasterScanTimeTol' that allow combining scans with different nominal time in a single radar volume. See merge_scans_odim in read_data_radar.py
Is that what you need or you also need something else?

wolfidan, 2023-04-20:
Oh wow amazing, this TimeTol should be what I'm looking for. Sorry for opening an issue but I thought it could be useful to have this referenced somewhere. I'm closing it now.

closed 2023-04-20 by wolfidan