MeteoSwiss/pyrad #51: Pyrad documentation issues

jfigui, 2023-08-23:
This issue relates to automatically generated [Pyrad documentation](https://meteoswiss.github.io/pyrad/).

In landing page:
1. Title. Current: "The Python ARM Radar Toolkit - Py-ART" should be "Pyrad"  
2. Development section: Github repository link not well defined
3. Citing Pyrad section: List of papers to cite should be bulleted
4. Disclaimer: Should we remove the "software still in development stage"?

in pyrad.io.get_data => keywords for datatypedescr not well rendered. We should probably display this information in Features section

List of Pyrad products => Some dataset families generate their own specific products but they can also generate all products of the VOL or GRID families. This is not reflected in the current documentation (e.g. (QVP product family is empty)

wolfidan, 2023-08-23:
Thanks @jfigui , I already fixed points 1-3, for the others I will try to get them fixed until sunday

wolfidan, 2023-08-24:
I fixed all points except :

in pyrad.io.get_data => keywords for datatypedescr not well rendered. We should probably display this information in Features section

This is a bigger job, indeed I think listing all possible input data types (RAINBOW, NEXRAD; COSMO, etc) should also be in the features section. But I still need to write the parsing code...

jfigui, 2023-12-12:
@wolfidan , I close this issue since all points have been addressed.

closed 2023-12-12 by jfigui