MeteoSwiss/pyrad #52: Py-ART documentation issues

jfigui, 2023-08-23:
This issue relates to the automatically generated [MeteoSwiss Py-ART documentation](https://github.com/MeteoSwiss/pyrad/issues/new)

Landing page:
1. Main title => should be "The MeteoSwiss Py-ART (Python ARM Radar Toolkit)"
2. Remove or update Developer Guide
3. Citing Py-ART => papers should be bulleted
4. List of Short Courses => Remove or refer to the last two where we were present
5. Getting help should point to either the discourse page or say that to get help you have to open an issue

User guide page:
1. Contributor's Guide should point to our own information on how to contribute.
2. Installation should point to our own information
3. Settin up an Environment should point to our own information and mention mini-forge instead of Anaconda

Changelog:
This should be our own changelog, not ARM Py-ART

jfigui, 2023-12-12:
@wolfidan, I think all points have been addressed except that in Installation section git clone gives the link to the ARM-DOE instead of the MeteoSwiss version. Can you change that and close the issue?

wolfidan, 2023-12-12:
@jfigui , fixed in b13a245

closed 2023-12-12 by wolfidan