MeteoSwiss/pyrad #53: Readme in main and test branches

jfigui, 2023-08-23:
This is a recurring issue:

For both Pyrad and Py-ART we have two branches: dev and master. Developments should target the dev branch whereas the master branch is only intended for releases.

On the readme in the dev branch there should be the text:

This branch is intended for MeteoSwiss Py-ART developers. Pull requests should target this branch!

And similar text for the Pyrad branch.

But this text SHOULD NOT be placed in the readme of the master branches.

Very often when pushing changes of the dev branch into the master branch we inadvertendly change the readme as well.

wolfidan, 2023-08-24:
Ciao @jfigui , I googled a bit and the way we are doing it is not considered good practice. It would make more sense imho to have the same readme for dev and master, and simply to state somewhere on the readme + user doc that pull requests should target the dev branch.

jfigui, 2023-12-12:
Hi @wolfidan , if that is the usual practice let's stick to it. Can you make the changes accordingly and close this issue?

wolfidan, 2023-12-19:
Closing the issue as it is fixed

closed 2023-12-19 by wolfidan