MeteoSwiss/pyrad #54: Problems when compiling from source code

jfigui, 2023-09-15:
Hi @wolfidan ,

I have recently re-installed pyrad from source and I have found some issues when compiling pyart using the scripts make_pyart.sh, or make_pyrad.sh.
When importing pyart it complains that it cannot find the following files:
/pyart/graph/chase-spectral-rgb.txt
/pyart/graph/spectral-extended-rgb.txt
/pyart/testing/registry.txt
They should be present in the build folder e.g. .local/lib/python3.9/site-packages/pyart_mch-1.8.3-py3.9-linux-x86_64.egg/

If I copy them manually everything is fine.

Do you know what those text files are about? I don't recall creating them. What would be the solution to copy them to the right place automatically?

wolfidan, 2023-09-15:
Ciao @jfigui ,

These are new colorbars that are used by ARM.
Sorry about that, normally pyart should include the files as it has a recursive include of .txt files in the MANIFEST.in.
I personally did not encounter this issue. To be on the safe side, I have added these files explicitely in the MANIFEST.in in commit [[BUG] added missing txt files in MANIFEST](https://github.com/MeteoSwiss/pyart/commit/c74d233b012ba83b7197c6d693a101446b7a5125) . Maybe you could test again and tell me. Thanks,

Cheers

jfigui, 2023-09-15:
Hi @wolfidan ,

It works flawless now. Thanks for the quick reaction. I close the issue.

Cheers

closed 2023-09-15 by jfigui