MeteoSwiss/pyrad #55: Not able to fetch the data

margish100, 2023-09-19:
hi @wolfidan,
yes i have using this new repo, but i have facing the same issue.

/Desktop/pyrad/pyrad/src/pyart/examples/plotting$ python3 plot_ppi_mdv.py

Traceback (most recent call last):
File "plot_ppi_mdv.py", line 17, in
from pyart.testing import get_test_data
ImportError: cannot import name 'get_test_data'

wolfidan, 2023-09-19:
Dear @margish100 ,

Thanks for your feedback, I will take a look into it today,
Cheers

margish100, 2023-09-20:
hello @wolfidan ,

main_process_data.py - This file also doesn't worked 

python3 main_process_data.py 'rad4alp_hydro_PLA_loc.txt' --starttime '20140523000000' --endtime '20140523001000' --postproc_cfgfile 'rad4alp_hydro_PLA_loc.txt' --cfgpath '$home/developer/Desktop/pyrad/pyrad/config/processing/'

Traceback (most recent call last):
  File "main_process_data_rt.py", line 47, in <module>
    from pyrad.flow.flow_control import main_rt as pyrad_main
  File "/home/developer/miniforge3/envs/pyrad/lib/python3.6/site-packages/pyrad/__init__.py", line 34, in <module>
    from . import proc
  File "/home/developer/miniforge3/envs/pyrad/lib/python3.6/site-packages/pyrad/proc/__init__.py", line 259, in <module>
    from .process_aux import get_process_func, process_raw, process_save_radar
  File "/home/developer/miniforge3/envs/pyrad/lib/python3.6/site-packages/pyrad/proc/process_aux.py", line 34, in <module>
    from pyart.util import cut_radar, compute_directional_stats
ImportError: cannot import name 'cut_radar'

wolfidan, 2023-09-21:
Dear @margish100 , 

sorry for the delay, I attended a workshop.
Could you please give me the output of 

`import pyart`
`print(pyart.__version__)`

margish100, 2023-10-03:
@wolfidan Thanks for the update. I will check and update you.
Thanks

wolfidan, 2023-12-19:
Hi @margish100 , do you have any news about this issue?

wolfidan, 2024-03-06:
Closing hte issue, as we didn't get any news

closed 2024-03-06 by wolfidan