MeteoSwiss/pyrad #61: Modification in get_sensor_data

jfigui, 2024-03-15:
Hi @wolfidan,

For info, I made a small modification in they way to get rain gauge data in function get_sensor_data. Now it first try to read the data using the function read_smn (as before) but if it does not succeed it tries again using the function read_smn2.

read_smn2 reads csv files with structure sensorid,time_stamp,value instead of the whole dataset from SwisMetNet stations. This is a simple hack that allows using RG data from any source.

The commit is this one: 2e89dea47483b95299ac0dd1b1c5a27762fe47b2

Let me know if you have any issue with that.

jfigui, 2024-06-17:
Hi @wolfidan ,

I noticed that you removed my modification in this commit 41382257dc5578d5cb121de11b3766b70e4c1878.

Was that intentional? In such case can you propose another solution? Because I need to be able to use read_smn and read_smn2 transparently

wolfidan, 2024-06-17:
Hi @jfigui, ah no, sorry about that it was not intentional. I tried to fix a deprecation warning that led to a crash on later versions of pandas and apparently I reverted the commit. I will fix it asap.

wolfidan, 2024-06-17:
Should be fixed in https://github.com/MeteoSwiss/pyrad/commit/f4fce2e2c798842b99463f8fa848b43992df349e

wolfidan, 2024-06-24:
@jfigui , can I close it?

closed 2024-09-30 by wolfidan